colloqi/piSignage #237: piSignage 5.0.1 with RPi.GPIO v0.7.1 not working with Pi 5

SCAI-Factory, 2024-01-11:
Due to the new Raspberry Pi 5 architecture.  The RPi.GPIO library is not compatible:  https://forums.raspberrypi.com/viewtopic.php?t=359742#:~:text=Any%20library%20written%20for%20the%20GPIO%20of%20earlier,You%20can%20not%20use%20RPi.GPIO%20on%20the%20Pi5.

A new method to interface/interface with the GPIO header pins will need to be developed for the https://github.com/colloqi/piSignage/blob/master/GPIO%20media%20control%20Instructions.md

SCAI-Factory, 2024-01-11:
Now that I think about this, this may only really affect me and anyone else using the RPi.GPIO library, so it might not be that big of a deal.

mano1979, 2024-03-21:
I too have issues with the gpio media control buttons. Any update on this issue? is there a fix?