colloqi/piSignage #239: Video Playback on 4k tvs very choppy

Noddley, 2024-01-13:
No matter what I do or how I format videos, videos in my playlist will be choppy or may not even play on 4K Sony TVs, even if they are at their native resolution or downscaled. The communication server is an open-source server. the there are Rassberry PI 4  are 8gb and 4gb

colloqi, 2024-01-14:
Hi

Note sure which version you are using on Pi, can you use 1080p settings under Group settings and try? Otherwise please write to us at support at pisignage.com with your account & player name.

Noddley, 2024-01-18:
Switching our raspberry pi 4 to 1080p worked! but playing 4k video is still broken even with VLC player. This is already a issue so I’m closing this branch thanks for your help

closed 2024-01-18 by Noddley