colloqi/piSignage #242: Left on desktop on fresh install 5.0.5 on pi4

tunisiano187, 2024-02-15:
Hello, just after installing the 5.0.5, starting the pi, just open the gnome (i think) desktop, no software seems installed or starting

AbhijithBn, 2024-02-19:
Hi @tunisiano187 

Did you use the pisignage 5.0.5 image from [HERE](https://github.com/colloqi/piSignage#getting-the-player-ready)?

Can you check the following?
1. Check if start.sh is present your home directory, and check the PID of the command using `ps -ef` command
2. Check if pm2 is running using the command `pm2 list`.
3. If pm2 is running, can you send us the logs [[Reference article](https://help.pisignage.com/hc/en-us/articles/360001255651-Player-logs)]

AbhijithBn, 2024-02-19:
The first boot usually takes some more time compared to subsequent boots and also might restart 1 or 2 times due to configuration changes.

tunisiano187, 2024-03-21:
[image]

tunisiano187, 2024-03-21:
Hello,
Forever out and Forevererr are empty both

mano1979, 2024-04-19:
This seems to happen if you use pi imager to add the wifi credentials/SSH credentials. Burn the image without any modification.

tunisiano187, 2024-04-24:
i'm not, i'm using win32DiskImager directly