colloqi/piSignage #243: No clock (pi 3 / debian buster / image 3.2.9)

itiser, 2024-02-20:
Hello!

I have the problem that on our **upgraded or newly installed** pi's no clock is shown although it is checked in the group options.

On all other players (pi 3 / debian stretch / image 3.2.9) the clock is shown (i upgraded them to 3.2.9 last week and updated 
onpremises pisignage server to the actual version).

3 of them are now installed with debian buster (2: fresh install / 1: os upgrade from stretch) and all of them do not show the clock as before.

Any Idea?

itiser, 2024-02-21:
After further research I saw that the clock is shown on TV until the video starts to play (after a reboot)....

itiser, 2024-03-12:
Any idea to solve this?

AbhijithBn, 2024-03-12:
Can you try using the Default Video player from Group settings?

itiser, 2024-03-12:
These are my settings:

[image]

[image]

So I'm using the default video player already...

itiser, 2024-04-08:
Nothing?

itiser, 2024-06-26:
Hello-oh!?

colloqi, 2024-06-27:
What is happening is clock is shown in browser and mostly video is shown by omxplayer or vlc which runs on top. To better understand your scenario, please write to support with account name/player name and we will have a look