colloqi/piSignage #244: Raspberry Pi 3A+ not booting

richland007, 2024-02-26:
Okay the strangest thing is happening  
I have a total of 4 Raspberry Pi 3A+ trying to build some menu signs for my restaurant and I must say that I am pretty handy with everything hi-tech.
I programmed the first card with pisingnage 3.2.0 a few days ago stuck it in one of my 3A+ pi's hooked it up to a display and it has been working flawlessly since than. 
After a few days I got the other 3 RPi's 3A+  plus 3 different micro sd cards and 3 different power supplies 3 different displays, 2 sets of software 3.2.0 and 3.2.9 and 2 different laptops with Balena etcher and win32diskimager... i have tried any and all combination possible to rule out something physical going bad from the pis itself to the cards to the displays to the software writing, power supplies and the laptops and none of them want to boot...
I can install regular Raspbian raspberry pi os on those sc cards and stick them to the 3A+ and they will boot and work just fine as expected.... just no pisignage....
On the other hand I also got any of the pissignage written sd cards that will not boot on the 3A+ and stick it to a regular raspberry pi 3B+ or a RPi Zero 2 W and it will work just fine...just not the other way around on the 3A+

Are there any power issues or display issues only pertaining to RPi 3A+??

What am I doing wrong why did one work and none others are working is there a know issue on the 3A+ that I am missing ....any help is appreciated since I do not have much time for this project,

TIA

jademedia1, 2024-04-11:
Did you get any luck on this at all? I have just received 2 brand new 3 A+ hardware units and am now experiencing the exact same issue.

jademedia1, 2024-04-11:
I have just realised that the 2018 PCB and the 2022 PCB have different components around the HDMI interface, so wondering if it is a board build issue creating this. I've got heaps of these players running flawlessly for the most part until I just bought 2 new boards and neither of them will output video for PiSignage OS, but will run fine on Raspbian.

jademedia1, 2024-04-11:
I have just realised that the 2018 PCB and the 2022 PCB have different components around the HDMI interface, so wondering if it is a board build issue creating this. I've got heaps of these players running flawlessly for the most part until I just bought 2 new boards and neither of them will output video for PiSignage OS, but will run fine on Raspbian.

[image]

jademedia1, 2024-04-14:
@colloqi Looking at revision notes from Raspberry Pi (https://pip.raspberrypi.com/categories/816-pcn) the revision only changed a power management chip. Perhaps there is some firmware difference that isn't accounted for in the revision notes? The guys at Core Electronics here in Australia have confirmed the same thing as I have, in that the 2018 board revision works flawlessly, but the 2022 revision doesn't.

Would this likely be a hard fix?