colloqi/piSignage #245: Mixing playlists

Mkragholm, 2024-02-26:
I am looking to mix playlists.
I am running a squash club and want to run 10-12 piSignage's - 2 in lounge areas, and 1 by each of 10 courts.
I have 1 playlist with various information that I will continuously update.
For each court I have 1 playlist consisting of just 1 weblink to todays bookings on that court. Similarly I have a playlist with a weblink to todays bookings across all courts for the lounge areas.

For all players I now want to run the Information playlist and the corresponding court/lounge playlist.
Ideally I would like the booking playlist to be shown as every second asset.
So if my Information playlist is Playlist "A" consisting of 3 assets "A1", "A2" and "A3", and I have my booking playlist as Playlist "B" consisting of only asset "B1", then I would like to show "A1", "B1", "A2", "B1", "A3", "B1", "A1", and so forth.

Is this possible, and if so - how do I set it up??

hitesh83, 2024-02-27:
two ways you can do this
1- make only one playlist per court with A1 B1 A2 B1 A3 B1 ......

2- make a Main playlist with B1 in it and 2nd playlist (Playlist A) as an advert playlist with A1 A2 A3 ...., you can schedule this 2nd playlist (Playlist A) to be played every desired second, and each time only one content of the playlist will be played and in the next cycle next item of Playlist A will be played.

Mkragholm, 2024-02-27:
@hitesh83 Thanks.
Method 1 would be annoying - every time I need to update an A-item I need to do it 12 times!!

But I think you may have nailed it with Method 2 - I will give that a shot!

Mkragholm, 2024-03-01:
@hitesh83 You were absolutely right! The Method 2 works like a charm! :-)

closed 2024-03-07 by Mkragholm