colloqi/piSignage #248: Requires manual start - New build + updates

probso, 2024-03-11:
With a fresh image piSignage does not autostart on a RPi4, completing all OS and piSignage updates it still fails to autostart.
The only way to get piSignage running is from a shell launch start.sh

Please can you provide instructions on how to setup such that the player auto-launches.

wrldtvlr, 2024-03-11:
Until the install script is updated, you can open an ssh terminal , then sudo raspi-config and set the auto login to GUI.

probso, 2024-03-11:
The GUI is starting and logging in, however I then need to manually open a terminal session and run start.sh due to piSignage not starting automatically.

colloqi, 2024-03-12:
Did you use the latest install scripts? Also which OS you are trying and can you use available image at https://github.com/colloqi/piSignage?tab=readme-ov-file#getting-the-player-ready instead?

AbhijithBn, 2024-04-22:
Hi @probso @wrldtvlr 

Please do not apply WiFi, Autologin, SSH customisations while installing pisignage image using Raspberry Pi Imager.
We recommend using Balena etcher if you are having this issue.