colloqi/piSignage #251: Player will stop working everyday

Buzzmovies, 2024-03-11:
Hello, every day, we get this image on our PiSignage TV.

[image]

Running 5.1.1 on a  Raspberry Pi 4 Model B Rev 1.1,  4GB

Same issues on 5.0.5, 5.1.0

When I "Sudo Reboot" it reloads fine but will stop after a while.

What I tried:
- Had the issue on WIFI, so I ran a wire
- Changed the Raspberry pi board
- Genuine Raspberry pi Power Supply
- Changed the MicroSD card 3 times

I dont know what to try anymore. Any assistance will be appreciated.

colloqi, 2024-03-12:
Hi

Can you try upgrading to 5.1.1 and run OS update as follows

sudo apt update; sudo apt -y full-upgrade &

If the problem still persists, let us know the account number at support at pisignage.com?

Thanks

AbhijithBn, 2024-04-22:
Hi @Buzzmovies 

This is fixed in the 5.1.4 release. 
Please update your player and let us know if you still face this issue.

Connexiumlabs, 2024-04-23:
@AbhijithBn Thanks for the heads up. We had to downgrade to an older version. If the issue persist after upgrade, I will follow up.
Thanks

Buzzmovies, 2024-04-23:
@AbhijithBn Thanks for the heads up. We had to downgrade to an older version. If the issue persists after the upgrade, I will follow up.
Thanks

closed 2024-04-23 by Buzzmovies