colloqi/piSignage #252: ! Important: loginscreen after ~30min -> piSignage not usable anymore!

Quietscher, 2024-03-15:
[image]
Morning!
After about 30 min, the greeter appears and is overlaid on top of piSignage.
Therefore, the Signage is not usable anymore.

System:
- Raspberry Pi 5
- Player Version 5.1.1
- Already did: `sudo apt update; sudo apt -y full-upgrade`

More information:
It seems to be the same problem as in [ISSUE](https://github.com/colloqi/piSignage/issues/251), but the issue's naming is misleading.
We tried to solve the issue by ourselves but couldn't fix it.
We found out:
- It is the greeter that is shown
- It may be a problem of the switch to Wayland / Wayfire
            -> On the internet there are dozens of solutions for X11 but barely for Wayland
           -> the configs (LightDm, freedesktop, wayfire.ini ...) seemed to be already set correctly, but still the greeter appears
- It may be that the greeter is shown because of inactivity

Resources we found, but that couldn't solve the problem:
- [https://pimylifeup.com/raspberry-pi-kiosk/](https://pimylifeup.com/raspberry-pi-kiosk/)
- [https://raspberrypi.stackexchange.com/questions/83041/where-is-raspberry-pi-logout-after-inactivity-configuration](https://raspberrypi.stackexchange.com/questions/83041/where-is-raspberry-pi-logout-after-inactivity-configuration)
- [https://github.com/WayfireWM/wayfire/wiki/Configuration#idlescreensaver_timeout](https://github.com/WayfireWM/wayfire/wiki/Configuration#idlescreensaver_timeout)
- [https://raspberrypi.stackexchange.com/questions/752/how-do-i-prevent-the-screen-from-going-blank](https://raspberrypi.stackexchange.com/questions/752/how-do-i-prevent-the-screen-from-going-blank)

<br>
<br>
- Further to this topic [https://forums.raspberrypi.com/viewtopic.php?p=2175546#p2175546](https://forums.raspberrypi.com/viewtopic.php?p=2175546#p2175546)

Please let me know if other information is required or if you found a fix.

Quietscher, 2024-03-15:
# These are the processes that run when the login screen appears

```
root     21804  0.0  0.1  29532  6272 ?        Sl   16:08   0:00 lightdm --session-child 18 21
lightdm  21891  0.0  0.2  16184  8704 ?        Ss   16:08   0:00 /lib/systemd/systemd --user
lightdm  21892  0.0  0.0  36304  3896 ?        S    16:08   0:00 (sd-pam)
lightdm  21911  0.0  0.1  48276  6144 ?        Ssl  16:08   0:00 /usr/bin/pipewire
lightdm  21912  0.0  0.3 106032 14392 ?        S<sl 16:08   0:00 /usr/bin/pulseaudio --daemonize=no --log-target=journal
lightdm  21913  0.1  4.0 384564 164864 ?       Ssl  16:08   0:00 /usr/bin/wayfire -c /usr/share/greeter.ini
lightdm  21918  0.0  0.0   7392  3712 ?        Ss   16:08   0:00 /usr/bin/dbus-daemon --session --address=systemd: --nofork --nopidfile --systemd-activation --syslog-only
lightdm  21973  0.0  0.0   2140  1152 ?        S    16:08   0:00 /bin/sh -c /usr/sbin/pi-greeter
lightdm  21974  0.2  3.9 268988 162700 ?       Sl   16:08   0:00 /usr/sbin/pi-greeter
lightdm  21982  0.0  0.1  44004  6272 ?        Ssl  16:08   0:00 /usr/libexec/gvfsd
lightdm  21987  0.0  0.1  56472  5632 ?        Sl   16:08   0:00 /usr/libexec/gvfsd-fuse /run/user/106/gvfs -f
root     22005  0.0  0.1  11040  4736 ?        S    16:08   0:00 lightdm --session-child 11 15
```

Quietscher, 2024-03-21:
Is there any new information on this issue yet?

mano1979, 2024-03-25:
> Is there any new information on this issue yet?

Have you burned the image to the sd card with pi imager? If so, did you use the extra settings to add wifi and ssh credentials while burning the image?

I did that too and had to log in each time the pi booted. But after doing a fresh install, (download the image again because the already present image is now corrupted), the pi logs in automatically.

Quietscher, 2024-03-25:
Hey @mano1979 👋 
No we used Balena Etcher, but I don't think that's the problem. We don't have any issues with ssh, wifi or login, but rather the raspberry / piSignage decides to **logout** after about 30min.

Quietscher, 2024-03-25:
I even wrote a script that simulates user input via pressing keys and moving the mouse. The tool is called **_ydotool_**.
But it does not work either

Quietscher, 2024-03-25:
@mano1979 do you have the newest piSignage version running? And do you get the logout screen after a certain time?

Quietscher, 2024-03-25:
Okay new Info:
**It does not seem to be a issue of inactivity**:
- a local script to simulate activity was running without effect
- real input over a connected mouse also had no effect

in both cases the login screen appears after 30 min.

colloqi, 2024-03-25:
Hi

We have observed when weblinks are used session getting closed and hence it goes back to login screen as bookworm has 2 users account (pi & root). We are actively monitoring the issue and working. Sometime the issue appears when you aplly the following commands (first one is to bring the OS up to date abd second one is to indicate HDMI is always connected)

Can you try if these help you?

 sudo apt update && sudo apt -y full-upgrade & 
sudo sed -i 's/$/ vc4.force_hotplug=1/' /boot/firmware/cmdline.txt

Quietscher, 2024-03-25:
[image]

When using
> sudo apt update && sudo apt -y full-upgrade & sudo sed -i 's/$/ vc4.force_hotplug=1/' /boot/firmware/cmdline.txt

The content is displayed outside of the monitor and the monitor only shows "piSignage"

Quietscher, 2024-03-25:
And when i do sudo sed -i 's/$/ **_vc4.force_hotplug=0_**/' /boot/firmware/cmdline.txt
the login screen still appears after 30min

colloqi, 2024-03-25:
It needs reboot after the command, if it still persists, till we solve, can you use 5.1.0-legacy if it is Pi 4?

Quietscher, 2024-03-25:
Yes i did reboot it in between.
I have a pi 5 with v5.1.1

Quietscher, 2024-03-25:
> <img alt="Bildschirmfoto 2024-03-25 um 11 13 12" width="585" src="https://private-user-images.githubusercontent.com/76123025/316465252-e0252111-9c28-47dc-a8bc-ef5e718fa95f.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTEzNjU0MTMsIm5iZiI6MTcxMTM2NTExMywicGF0aCI6Ii83NjEyMzAyNS8zMTY0NjUyNTItZTAyNTIxMTEtOWMyOC00N2RjLWE4YmMtZWY1ZTcxOGZhOTVmLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAzMjUlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMzI1VDExMTE1M1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTVlM2FmM2ExYzA4MjE3MTY0YTY2YmRkMjgzODUwZTNjNmMyZTcxODQwMTg4YzRlMTc1NzE1ODhkZmQwYTM1NDYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.5yYRo50GkbAK6kENIicPnv3csEaiZm9oFSHq849wIIM">
> When using
> 
> > sudo apt update && sudo apt -y full-upgrade & sudo sed -i 's/$/ vc4.force_hotplug=1/' /boot/firmware/cmdline.txt
> 
> The content is displayed outside of the monitor and the monitor only shows "piSignage"

Is there a fix for this?
This seems to work against the login screen, but as i wrote, the display does not display my content but piSignage

Quietscher, 2024-03-25:
Ah nope my bad, id did not work either

Quietscher, 2024-03-26:
> Okay new Info: **It does not seem to be a issue of inactivity**:
> 
> * a local script to simulate activity was running without effect
> * real input over a connected mouse also had no effect
> 
> in both cases the login screen appears after 30 min.

I wonder why and where this comes from...

Mkragholm, 2024-03-26:
This is very similar to the issue I had where it needed a password on every reboot.
That issue is solved in Issue #246 - give that a shot.

Quietscher, 2024-03-26:
Thanks for the info, but that did not help. I tried it already

Quietscher, 2024-03-26:
I don't get the sign in screen on reboot, but after 30min

stormvansoldt, 2024-03-30:
> # These are the processes that run when the login screen appears
> 
> ```
> root     21804  0.0  0.1  29532  6272 ?        Sl   16:08   0:00 lightdm --session-child 18 21
> ```

I'm really curious where this process is being generated from. Maybe check the crontab? 

Does lightdm have any useful messages in /var/log?

Quietscher, 2024-04-02:
[+0.00s] DEBUG: Logging to /var/log/lightdm/lightdm.log
[+0.00s] DEBUG: Starting Light Display Manager 1.26.0, UID=0 PID=863
[+0.00s] DEBUG: Loading configuration dirs from /usr/share/lightdm/lightdm.conf.d
[+0.00s] DEBUG: Loading configuration from /usr/share/lightdm/lightdm.conf.d/01_debian.conf
[+0.00s] DEBUG: Loading configuration dirs from /usr/local/share/lightdm/lightdm.conf.d
[+0.00s] DEBUG: Loading configuration dirs from /etc/xdg/lightdm/lightdm.conf.d
[+0.00s] DEBUG: Loading configuration from /etc/lightdm/lightdm.conf
[+0.00s] DEBUG: Registered seat module local
[+0.00s] DEBUG: Registered seat module xremote
[+0.00s] DEBUG: Registered seat module unity
[+0.00s] DEBUG: Using D-Bus name org.freedesktop.DisplayManager
[+0.00s] DEBUG: Using cross-namespace EXTERNAL authentication (this will deadlock if server is GDBus < 2.73.3)
[+0.01s] DEBUG: _g_io_module_get_default: Found default implementation local (GLocalVfs) for ‘gio-vfs’
[+0.01s] DEBUG: Monitoring logind for seats
[+0.01s] DEBUG: New seat added from logind: seat0
[+0.01s] DEBUG: Seat seat0: Loading properties from config section Seat:*
[+0.01s] DEBUG: Seat seat0 has property CanMultiSession=no
[+0.01s] DEBUG: Seat seat0: Starting
[+0.01s] DEBUG: Seat seat0: Creating user session
[+0.01s] DEBUG: Loading users from org.freedesktop.Accounts
[+0.01s] DEBUG: User /org/freedesktop/Accounts/User1000 added
[+0.32s] DEBUG: Seat seat0: Creating display server of type wayland
[+0.32s] DEBUG: Seat seat0: Plymouth is running on VT 1, but this is less than the configured minimum of 7 so not replacing it
[+0.32s] DEBUG: Quitting Plymouth
[+0.38s] DEBUG: Using VT 7
[+0.38s] DEBUG: Launching process 997: /usr/share/dispsetup.sh
[+0.39s] DEBUG: Process 997 exited with return value 0
[+0.39s] DEBUG: Seat seat0: Exit status of /usr/share/dispsetup.sh: 0
[+0.39s] DEBUG: Seat seat0: Display server ready, starting session authentication
[+0.39s] DEBUG: Session pid=1000: Started with service 'lightdm-autologin', username 'pi'
[+0.39s] DEBUG: Acquired bus name org.freedesktop.DisplayManager
[+0.39s] DEBUG: Registering seat with bus path /org/freedesktop/DisplayManager/Seat0
[+0.42s] DEBUG: Session pid=1000: Authentication complete with return value 0: Success
[+0.42s] DEBUG: Seat seat0: Session authenticated, running command
[+0.42s] DEBUG: Registering session with bus path /org/freedesktop/DisplayManager/Session0
[+0.42s] DEBUG: Session pid=1000: Running command /etc/X11/Xsession /usr/bin/wayfire-pi
[+0.42s] DEBUG: Creating shared data directory /var/lib/lightdm/data/pi
[+0.42s] DEBUG: Session pid=1000: Logging to .xsession-errors
[+0.78s] DEBUG: Activating VT 7
[+0.80s] DEBUG: Activating login1 session 1
[+0.80s] DEBUG: Seat seat0 changes active session to 1
[+0.80s] DEBUG: Session 1 is already active
[+2191.81s] DEBUG: Session pid=1000: Exited with return value 255
[+2191.82s] DEBUG: Seat seat0: Session stopped
[+2191.82s] DEBUG: Seat seat0: Stopping display server, no sessions require it
[+2191.82s] DEBUG: Seat seat0: Display server stopped
[+2191.82s] DEBUG: Seat seat0: Active display server stopped, starting greeter
[+2191.82s] DEBUG: Seat seat0: Creating greeter session
[+2192.13s] DEBUG: Seat seat0: Creating display server of type wayland
[+2192.13s] DEBUG: Using VT 8
[+2192.13s] DEBUG: Launching process 22405: /usr/share/dispsetup.sh
[+2192.14s] DEBUG: Process 22405 exited with return value 0
[+2192.14s] DEBUG: Seat seat0: Exit status of /usr/share/dispsetup.sh: 0
[+2192.14s] DEBUG: Seat seat0: Display server ready, starting session authentication
[+2192.14s] DEBUG: Session pid=22408: Started with service 'lightdm-greeter', username 'lightdm'
[+2192.14s] DEBUG: Releasing VT 7
[+2192.24s] DEBUG: Session pid=22408: Authentication complete with return value 0: Success
[+2192.24s] DEBUG: Seat seat0: Session authenticated, running command
[+2192.24s] DEBUG: Session pid=22408: Running command /usr/bin/wayfire-pi -c /usr/share/greeter.ini
[+2192.24s] DEBUG: Creating shared data directory /var/lib/lightdm/data/lightdm
[+2192.24s] DEBUG: Session pid=22408: Logging to /var/log/lightdm/seat0-greeter.log
[+2192.50s] DEBUG: Activating VT 8
[+2192.51s] DEBUG: Activating login1 session c1
[+2192.55s] DEBUG: Seat seat0 changes active session to c1
[+2192.55s] DEBUG: Session c1 is already active
[+2193.65s] DEBUG: Greeter connected version=1.26.0 api=1 resettable=false
[+2194.05s] DEBUG: Greeter start authentication for pi
[+2194.05s] DEBUG: Session pid=22622: Started with service 'lightdm', username 'pi'
[+2194.05s] DEBUG: Session pid=22622: Got 1 message(s) from PAM
[+2194.05s] DEBUG: Prompt greeter with 1 message(s)

Quietscher, 2024-04-02:
This is in the forever out log, when the greeter appears

2024-04-02T08:19:59: (chromium-browser:21847): Gtk-WARNING **: 08:19:59.379: Theme parsing error: gtk.css:5:116: 'height' is not a valid property name
2024-04-02T08:19:59: 
2024-04-02T08:19:59: (chromium-browser:21847): Gtk-WARNING **: 08:19:59.379: Theme parsing error: gtk.css:5:142: 'max-height' is not a valid property name
2024-04-02T08:19:59: [21847:21847:0402/081959.475262:ERROR:policy_logger.cc(157)] :components/enterprise/browser/controller/chrome_browser_cloud_management_controller.cc(161) Cloud management controller initialization aborted as CBCM is not enabled. Please use the `--enable-chrome-browser-cloud-management` command line flag to enable it if you are not using the official Google Chrome build.
2024-04-02T08:19:59: 
2024-04-02T08:19:59: DevTools listening on ws://127.0.0.1:46383/devtools/browser/6d9f1c5f-bfb3-40ba-a717-caa1fbeb6594
2024-04-02T08:19:59: [21847:21977:0402/081959.769976:ERROR:object_proxy.cc(576)] Failed to call method: org.freedesktop.DBus.Properties.Get: object_path= /org/freedesktop/UPower: org.freedesktop.DBus.Error.ServiceUnknown: The name org.freedesktop.UPower was not provided by any .service files
2024-04-02T08:19:59: [21847:21977:0402/081959.771155:ERROR:object_proxy.cc(576)] Failed to call method: org.freedesktop.UPower.GetDisplayDevice: object_path= /org/freedesktop/UPower: org.freedesktop.DBus.Error.ServiceUnknown: The name org.freedesktop.UPower was not provided by any .service files
2024-04-02T08:19:59: [21847:21977:0402/081959.771600:ERROR:object_proxy.cc(576)] Failed to call method: org.freedesktop.UPower.EnumerateDevices: object_path= /org/freedesktop/UPower: org.freedesktop.DBus.Error.ServiceUnknown: The name org.freedesktop.UPower was not provided by any .service files
2024-04-02T08:19:59: close object 1: Invalid argument
2024-04-02T08:19:59: close object 2: Invalid argument
2024-04-02T08:19:59: close object 3: Invalid argument
2024-04-02T08:19:59: close object 4: Invalid argument
2024-04-02T08:19:59: close object 5: Invalid argument
2024-04-02T08:19:59: close object 6: Invalid argument
2024-04-02T08:20:03: close object 1: Invalid argument
2024-04-02T08:20:03: close object 2: Invalid argument
2024-04-02T08:20:03: close object 3: Invalid argument
2024-04-02T08:20:04: close object 4: Invalid argument
2024-04-02T08:20:04: close object 5: Invalid argument
2024-04-02T08:20:04: close object 6: Invalid argument
2024-04-02T08:20:19: puppeteer close event for http://localhost/mywebsite: undefined
2024-04-02T08:20:19: Browser stopped with code 1 and signal null
2024-04-02T08:20:19: [22033:22033:0402/082019.501816:ERROR:wayland_connection.cc(196)] Failed to connect to Wayland display: Connection refused (111)
2024-04-02T08:20:19: [22033:22033:0402/082019.501884:ERROR:ozone_platform_wayland.cc(264)] Failed to initialize Wayland platform
2024-04-02T08:20:19: [22033:22033:0402/082019.501899:ERROR:env.cc(257)] The platform failed to initialize.  Exiting.
2024-04-02T08:20:19: Puppeteer launch exception Error: Failed to launch the browser process! undefined
2024-04-02T08:20:19: [22033:22033:0402/082019.501816:ERROR:wayland_connection.cc(196)] Failed to connect to Wayland display: Connection refused (111)
2024-04-02T08:20:19: [22033:22033:0402/082019.501884:ERROR:ozone_platform_wayland.cc(264)] Failed to initialize Wayland platform
2024-04-02T08:20:19: [22033:22033:0402/082019.501899:ERROR:env.cc(257)] The platform failed to initialize.  Exiting.
2024-04-02T08:20:19: 
2024-04-02T08:20:19: 
2024-04-02T08:20:19: TROUBLESHOOTING: https://pptr.dev/troubleshooting
2024-04-02T08:20:19: 
2024-04-02T08:20:19: *** puppeteer launch error, not proceeding

stormvansoldt, 2024-04-02:
Thanks for that output, I have a few PiSignage devices on the latest 5.X version and I haven't been able to re-create the issue.

What does your playlist look like/what type of content does it contain?

Quietscher, 2024-04-02:
It's just some webpages that are served on a locally running nginx. It is a compiled Vue project with routing.

stormvansoldt, 2024-04-02:
This is a REALLY stupid workaround and might not even solve the core issue, but here's what I'm testing right now:

```
sudo sed -i '/^autostart0/ s/./#&/' /etc/wayfire/greeter.ini
sudo reboot now
```

Basically just disabling the greeter altogether. I already confirmed the player still boots normally with this change and currently have it cycling through a few selfhosted web pages. I'll update once I hit the 30-minute timer

AbhijithBn, 2024-04-03:
Hi @Quietscher 

We are actively working on this issue.

Can you try the following?

1) In the file `/etc/lightdm/lightdm.conf`, search for the following lines, uncomment it and change the values accordingly

```
greeter-user=pi
pam-greeter-service=lightdm-autologin
```
This should autologin pi user and also prevent Chrome from throwing Invalid user/auth errors.

2) In the file `/etc/wayfire/greeter.ini`, comment the line

`#autostart0 = /usr/sbin/pi-greeter `

This should prevent the greeter UI from showing up.

Please let us know if this solves the issue. We will be testing the same too.

Thank you for all your previous debug logs and patience.

Quietscher, 2024-04-03:
Ok I'll first try @AbhijithBn approach, i will let you know about the results. Afterwards if the problem still occurs i will try your approach @stormvansoldt. Thanks for now!

Quietscher, 2024-04-03:
Also btw, can you add an option to set the cursor position? at the moment it is centered on the screen (but invisible) and therefore it triggers a hover effect on my website that i am showing, that should not occur.
Or is there an option to move it after startup (e.g. to a screen corner)

Quietscher, 2024-04-03:
Ok with @AbhijithBn Solution the screen now just turns black after 30min

Quietscher, 2024-04-03:
ah okay @stormvansoldt solution is the same as @AbhijithBn solution:
#autostart0 = /usr/sbin/pi-greeter 

I am sorry but it does not work as intended. the screensaver might be gone, but now the screen is just black by itself

Quietscher, 2024-04-03:
This is the log
[+0.00s] DEBUG: Logging to /var/log/lightdm/lightdm.log
[+0.00s] DEBUG: Starting Light Display Manager 1.26.0, UID=0 PID=865
[+0.00s] DEBUG: Loading configuration dirs from /usr/share/lightdm/lightdm.conf.d
[+0.00s] DEBUG: Loading configuration from /usr/share/lightdm/lightdm.conf.d/01_debian.conf
[+0.00s] DEBUG: Loading configuration dirs from /usr/local/share/lightdm/lightdm.conf.d
[+0.00s] DEBUG: Loading configuration dirs from /etc/xdg/lightdm/lightdm.conf.d
[+0.00s] DEBUG: Loading configuration from /etc/lightdm/lightdm.conf
**[+0.00s] DEBUG:   [Seat:*] contains unknown option greeter-user**
[+0.00s] DEBUG: Registered seat module local
[+0.00s] DEBUG: Registered seat module xremote
[+0.00s] DEBUG: Registered seat module unity
[+0.00s] DEBUG: Using D-Bus name org.freedesktop.DisplayManager
[+0.00s] DEBUG: Using cross-namespace EXTERNAL authentication (this will deadlock if server is GDBus < 2.73.3)
[+0.01s] DEBUG: _g_io_module_get_default: Found default implementation local (GLocalVfs) for ‘gio-vfs’
[+0.01s] DEBUG: Monitoring logind for seats
[+0.01s] DEBUG: New seat added from logind: seat0
[+0.01s] DEBUG: Seat seat0: Loading properties from config section Seat:*
[+0.01s] DEBUG: Seat seat0 has property CanMultiSession=no
[+0.01s] DEBUG: Seat seat0: Starting
[+0.01s] DEBUG: Seat seat0: Creating user session
[+0.01s] DEBUG: Loading users from org.freedesktop.Accounts
[+0.01s] DEBUG: User /org/freedesktop/Accounts/User1000 added
[+0.35s] DEBUG: Seat seat0: Creating display server of type wayland
[+0.35s] DEBUG: Seat seat0: Plymouth is running on VT 1, but this is less than the configured minimum of 7 so not replacing it
[+0.35s] DEBUG: Quitting Plymouth
[+0.43s] DEBUG: Using VT 7
[+0.43s] DEBUG: Launching process 971: /usr/share/dispsetup.sh
[+0.43s] DEBUG: Process 971 exited with return value 0
[+0.43s] DEBUG: Seat seat0: Exit status of /usr/share/dispsetup.sh: 0
[+0.43s] DEBUG: Seat seat0: Display server ready, starting session authentication
[+0.43s] DEBUG: Session pid=973: Started with service 'lightdm-autologin', username 'pi'
[+0.43s] DEBUG: Acquired bus name org.freedesktop.DisplayManager
[+0.43s] DEBUG: Registering seat with bus path /org/freedesktop/DisplayManager/Seat0
[+0.46s] DEBUG: Session pid=973: Authentication complete with return value 0: Success
[+0.46s] DEBUG: Seat seat0: Session authenticated, running command
[+0.46s] DEBUG: Registering session with bus path /org/freedesktop/DisplayManager/Session0
[+0.46s] DEBUG: Session pid=973: Running command /etc/X11/Xsession /usr/bin/wayfire-pi
[+0.46s] DEBUG: Creating shared data directory /var/lib/lightdm/data/pi
[+0.46s] DEBUG: Session pid=973: Logging to .xsession-errors
[+0.88s] DEBUG: Activating VT 7
[+0.90s] DEBUG: Activating login1 session 1
[+0.90s] DEBUG: Seat seat0 changes active session to 1
[+0.90s] DEBUG: Session 1 is already active
**[+2188.79s] DEBUG: Session pid=973: Exited with return value 255**
[+2188.79s] DEBUG: Seat seat0: Session stopped
[+2188.79s] DEBUG: Seat seat0: Stopping display server, no sessions require it
**[+2188.79s] DEBUG: Seat seat0: Display server stopped**
**[+2188.79s] DEBUG: Seat seat0: Active display server stopped, starting greeter**
[+2188.79s] DEBUG: Seat seat0: Creating greeter session
[+2189.09s] DEBUG: Seat seat0: Creating display server of type wayland
[+2189.09s] DEBUG: Using VT 8
[+2189.09s] DEBUG: Launching process 21709: /usr/share/dispsetup.sh
[+2189.09s] DEBUG: Process 21709 exited with return value 0
[+2189.09s] DEBUG: Seat seat0: Exit status of /usr/share/dispsetup.sh: 0
[+2189.09s] DEBUG: Seat seat0: Display server ready, starting session authentication
**[+2189.09s] DEBUG: Session pid=21711: Started with service 'lightdm-autologin', username 'lightdm'**
[+2189.09s] DEBUG: Releasing VT 7
[+2189.22s] DEBUG: Session pid=21711: Authentication complete with return value 0: Success
[+2189.22s] DEBUG: Seat seat0: Session authenticated, running command
**[+2189.22s] DEBUG: Session pid=21711: Running command /usr/bin/wayfire-pi -c /usr/share/greeter.ini**
[+2189.22s] DEBUG: Creating shared data directory /var/lib/lightdm/data/lightdm
[+2189.22s] DEBUG: Session pid=21711: Logging to /var/log/lightdm/seat0-greeter.log

→ the seat0-greeter.log log is empty

[+2189.54s] DEBUG: Activating VT 8
[+2189.55s] DEBUG: Activating login1 session 6
[+2189.57s] DEBUG: Seat seat0 changes active session to 6
[+2189.57s] DEBUG: Session 6 is already active

Quietscher, 2024-04-03:
I dont know if it crashed. The screen is just black but the cursor is there when a mouse is connected and i can move it

colloqi, 2024-04-03:
Looks like user is not changed to pi here

greeter-user=pi

as per this line

[+2189.09s] DEBUG: Session pid=21711: Started with service 'lightdm-autologin', username 'lightdm'

Quietscher, 2024-04-03:
yes it says:
[+0.00s] DEBUG: [Seat:*] contains unknown option greeter-user

Quietscher, 2024-04-03:
```
#
# General configuration
#
# start-default-seat = True to always start one seat if none are defined in the configuration
# greeter-user = User to run greeter as
# minimum-display-number = Minimum display number to use for X servers
# minimum-vt = First VT to run displays on
# lock-memory = True to prevent memory from being paged to disk
# user-authority-in-system-dir = True if session authority should be in the system location
# guest-account-script = Script to be run to setup guest account
# logind-check-graphical = True to on start seats that are marked as graphical by logind
# log-directory = Directory to log information to
# run-directory = Directory to put running state in
# cache-directory = Directory to cache to
# sessions-directory = Directory to find sessions
# remote-sessions-directory = Directory to find remote sessions
# greeters-directory = Directory to find greeters
# backup-logs = True to move add a .old suffix to old log files when opening new ones
# dbus-service = True if LightDM provides a D-Bus service to control it
#
[LightDM]
#start-default-seat=true
#greeter-user=lightdm
#minimum-display-number=0
#minimum-vt=7
#lock-memory=true
#user-authority-in-system-dir=false
#guest-account-script=guest-account
#logind-check-graphical=false
#log-directory=/var/log/lightdm
#run-directory=/var/run/lightdm
#cache-directory=/var/cache/lightdm
#sessions-directory=/usr/share/lightdm/sessions:/usr/share/xsessions:/usr/share/wayland-sessions
#remote-sessions-directory=/usr/share/lightdm/remote-sessions
#greeters-directory=$XDG_DATA_DIRS/lightdm/greeters:$XDG_DATA_DIRS/xgreeters
#backup-logs=true
#dbus-service=true

#
# Seat configuration
#
# Seat configuration is matched against the seat name glob in the section, for example:
# [Seat:*] matches all seats and is applied first.
# [Seat:seat0] matches the seat named "seat0".
# [Seat:seat-thin-client*] matches all seats that have names that start with "seat-thin-client".
#
# type = Seat type (local, xremote, unity)
# pam-service = PAM service to use for login
# pam-autologin-service = PAM service to use for autologin
# pam-greeter-service = PAM service to use for greeters
# xserver-backend = X backend to use (mir)
# xserver-command = X server command to run (can also contain arguments e.g. X -special-option)
# xmir-command = Xmir server command to run (can also contain arguments e.g. Xmir -special-option)
# xserver-config = Config file to pass to X server
# xserver-layout = Layout to pass to X server
# xserver-allow-tcp = True if TCP/IP connections are allowed to this X server
# xserver-share = True if the X server is shared for both greeter and session
# xserver-hostname = Hostname of X server (only for type=xremote)
# xserver-display-number = Display number of X server (only for type=xremote)
# xdmcp-manager = XDMCP manager to connect to (implies xserver-allow-tcp=true)
# xdmcp-port = XDMCP UDP/IP port to communicate on
# xdmcp-key = Authentication key to use for XDM-AUTHENTICATION-1 (stored in keys.conf)
# unity-compositor-command = Unity compositor command to run (can also contain arguments e.g. unity-system-compositor -special-option)
# unity-compositor-timeout = Number of seconds to wait for compositor to start
# greeter-session = Session to load for greeter
# greeter-hide-users = True to hide the user list
# greeter-allow-guest = True if the greeter should show a guest login option
# greeter-show-manual-login = True if the greeter should offer a manual login option
# greeter-show-remote-login = True if the greeter should offer a remote login option
[... 86 lines cut ...]
#hostname=

#
# VNC Server configuration
#
# enabled = True if VNC connections should be allowed
# command = Command to run Xvnc server with
# port = TCP/IP port to listen for connections on
# listen-address = Host/address to listen for VNC connections (use all addresses if not present)
# width = Width of display to use
# height = Height of display to use
# depth = Color depth of display to use
#
[VNCServer]
#enabled=false
#command=Xvnc
#port=5900
#listen-address=
#width=1024
#height=768
#depth=8

```

**This is the current lightdm config**

AbhijithBn, 2024-04-03:
[image]

colloqi, 2024-04-03:
autologin-user=pi

Quietscher, 2024-04-03:
ah okay, ill do that

Quietscher, 2024-04-03:
Ok now after 30min:
My pages are still shown **but**:
It does not apply the 1080p settings and my screen is 4k, so its now only covering 1/4 of the screen (the top left corner).
But it should scale the 1080p content up to 4k, as it does on a fresh boot of the piSignage

Quietscher, 2024-04-03:
[image]

Quietscher, 2024-04-03:
Updates:
It is a cycle:
after 30min it is displayed only in the top left corner as 1/4 of the screen size
after the next 30 min it is displayed as it should
then it repeats.

AbhijithBn, 2024-04-03:
Can you add this code to the `/usr/share/greeter.ini` file?

```
[output]
[output:HDMI-A-1]
mode = 1920x1080
transform = normal
```

Your file should look like this
```
[input-device:10-0038 generic ft5x06 (79)]
output = DSI-1

[input-device:6-0038 generic ft5x06 (79)]
output = DSI-1

[input-device:4-0038 generic ft5x06 (79)]
output = DSI-2

[input-device:10-0038 generic ft5x06 (00)]
output = DSI-1

[input-device:6-0038 generic ft5x06 (00)]
output = DSI-1

[input-device:4-0038 generic ft5x06 (00)]
output = DSI-2

[input]
xkb_options=
xkb_model=pc105
xkb_layout=us
xkb_variant=

[output]
[output:HDMI-A-1]
mode = 1920x1080
transform = normal

```

Quietscher, 2024-04-03:
ok!

AbhijithBn, 2024-04-03:
Additionally to remove the cursor, you can modify the `/etc/wayfire/greeter.ini`. You can include the **hide-cursor** plugin under [core] section.

```
plugins = autostart-static pixdecor winshadows hide-cursor
plugins_nogl = autostart-static pixdecor hide-cursor
```

Quietscher, 2024-04-03:
> Additionally to remove the cursor, you can modify the `/etc/wayfire/greeter.ini`. You can include the **hide-cursor** plugin under [core] section.
> 
> ```
> plugins = autostart-static pixdecor winshadows hide-cursor
> plugins_nogl = autostart-static pixdecor hide-cursor
> ```

i can try this later, i hope it does not trigger hover effects. Is there a possibility to just move it via default to a corner? e.g. top left?

AbhijithBn, 2024-04-03:
No it does not trigger hover effects.
Currently when the piSignage player starts playback the mouse cursor is hidden and we would like to maintain that consistency.

Quietscher, 2024-04-03:
> Yes, you can try it later. No it does not trigger hover effects. Currently when the piSignage player starts playback the mouse cursor is hidden and we would like to maintain that consistency.

Ah i understand. I will try it

Quietscher, 2024-04-03:
> Can you add this code to the `/usr/share/greeter.ini` file?
>
> ```
> [output]
> [output:HDMI-A-1]
> mode = 1920x1080
> transform = normal
> ```
>
> Your file should look like this
>
> ```
> [input-device:10-0038 [...]

No I am sorry, it did not help

Quietscher, 2024-04-03:
> i can try this later, i hope it does not trigger hover effects. Is there a possibility to just move it via default to a corner? e.g. top left?

Also this does not work
The cursor is not visible, but it is at the center of the screen and triggers hover effects on the websites

AbhijithBn, 2024-04-03:
Can you share the contents of the file /usr/share/greeter.ini ?

Quietscher, 2024-04-03:
[image]

AbhijithBn, 2024-04-03:
Can you share the output of the command `wlr-randr`?

Quietscher, 2024-04-03:
It says HDMI-A-2
I added:
```
[output]
[output:HDMI-A-0]
mode = 1920x1080
transform = normal

[output:HDMI-A-1]
mode = 1920x1080
transform = normal

[output:HDMI-A-2]
mode = 1920x1080
transform = normal
```

Quietscher, 2024-04-03:
I'll let you know tomorrow if it works

Quietscher, 2024-04-04:
Ok It seems to be fixed. The only thing left is that the cursor is still triggering hovereffects.

AbhijithBn, 2024-04-04:
That's Great !

Quietscher, 2024-04-04:
Any news about the cursor?

AbhijithBn, 2024-04-04:
We'll update this space once we come up with anything.

AbhijithBn, 2024-04-05:
Hi @Quietscher 

Can you use the latest 5.1.2 piSignage image [LINK](https://pisignage.s3.amazonaws.com/pisignage-images/pisignage_5.1.0.img.zip) ?

Quietscher, 2024-04-08:
@AbhijithBn Did you change anything regarding the cursor?

AbhijithBn, 2024-04-08:
No, but we have a different approach to the LightDM crash issue. This should bypass the cursor issue mentioned.

[56 more replies]