colloqi/piSignage #256: New image for rpi3?

itiser, 2024-04-08:
Is there new image planned for rpi3 based on bullseye/bookworm?

EOL is coming...

[image]

colloqi, 2024-04-10:
We are waiting for the optimized bookworm release for Pi 3 , till then you can try 5.1.0-legacy if needed

jademedia1, 2024-04-12:
Is there any chance the new  image when done can be checked on the 3a+ boards as well - I am getting no HDMI output on the newer 2022 version boards, but 5.1.0-legacy at least gave a few seconds output and prompted to tv for the correct screen resolution first before hanging.

itiser, 2024-04-22:
> We are waiting for the optimized bookworm release for Pi 3 , till then you can try 5.1.0-legacy if needed

[image]

??

AbhijithBn, 2024-04-22:
@itiser 
3.2.9 is the recommended piSignage version for Pi 3.
If you are concerned by EOL, you can try the 5.1.0-legacy version of pisignage but there might be some differences in the behaviour and performance.

jademedia1, 2024-04-24:
5.1.0-legacy doesn't work at all on the Pi3 as far as I have been able to test (neither A= or B would run it at all.

itiser, 2024-06-26:
Only 4 days left!

[image]

Since 5.1.0 doesn't work - what's your solution?