colloqi/piSignage #257: 5.1.2 getting stuck on Pi logo

funzoneq, 2024-04-09:
I upgraded my Raspberry Pi 5 via the portal to the latest version. It reboots. Pi logo shows up. Then it reboots again. Pi logo shows up and it never continues.

I tried downloading the binary from the github page, flashed it with BalenaEtcher and Pi Imager.

There is no shasum I can compare my image against. But I downloaded it both via S3 and Google Drive. Both have the same signature:

```
shasum -a 256 pisignage_5.1.2.img
2ef2f973b2e14280f7a19f60961eb302911b094beb4a8bf6b152a96bf604af94  pisignage_5.1.2.img
```

colloqi, 2024-04-10:
Hi, are you using any customization options of the Raspberry Pi imager, please do not use them? If possible, can you use ctrl+alt+t to get the terminal and send the content of forever_out.log and forever_err.log files?

funzoneq, 2024-04-10:
No options used with the imagers. I can ssh to the box and login.

Here is the requested log:
[forever_out.log](https://github.com/colloqi/piSignage/files/14933478/forever_out.log)

pi@Pi5-Kantoor:~ $ ls -al *.log
-rw-r--r-- 1 pi pi     1 Apr  5 10:29 forever_err.log
-rw-r--r-- 1 pi pi 22461 Apr 10 16:19 forever_out.log

Error log is empty.

funzoneq, 2024-04-10:
[image]

colloqi, 2024-04-10:
Looks like the cable is connected to HDMI-2, can you connect the HDMI cable immediately next to the power supply port?

`Available displays: [{"interfaceName":"HDMI-A-2","interfaceResolutionAndOffset":"1920x1080+1200+0","resolution":["1920","1080"],"offset":["1200","0"]},{"interfaceName":"HDMI-A-1","interfaceResolutionAndOffset":"+","resolution":[""],"offset":[""]}]`

funzoneq, 2024-04-10:
[forever_out.log](https://github.com/colloqi/piSignage/files/14935151/forever_out.log)

The player is now stuck on the Welcome - Player ID page. The player shows a yellow status in the dashboard. I tried to deploy the playlist to it, but that doesn't seem to do anything.

AbhijithBn, 2024-04-11:
Hi  @funzoneq 

TV_OFF was scheduled from Group Settings and hence no playlist was being played. We have disabled the TV_OFF and Reboot scheduled from Group settings. 
We recommend scheduling the TV_OFF playlist for finer control.

```
2024-04-10T18:45:51: Copied server files for the group default successfully.
2024-04-10T18:45:51: Stopping Lounge music
2024-04-10T18:45:51: TV is OFF, skipping playlist play
```

funzoneq, 2024-04-11:
Yeah, seems like it's working as expected now. So in the end it was just changing the HDMI port. Thanks for the support :)

closed 2024-04-11 by funzoneq