colloqi/piSignage #259: 2 MFA forced even if NONE selected

Buzzmovies, 2024-05-11:
[image]

Hello, we have issues where our "Collaborators" are forced to use 2MFA even though NONE were previously selected settings.

Basically, It's like 2 MFA is enforced for everyone by default even if showing NONE

Thank you

colloqi, 2024-05-12:
Hi, we are enforcing one time email verification for every account for each location of login to avoid unscrupulous logins. If collaborator does not have email it is sent to the account holder's email.

Buzzmovies, 2024-05-12:
@colloqi , Thank you for your reply. I now understand. 

Question: After one-time verification, is it possible to disable 2MFA or use an authenticator app if no email is associated with the collaborator's account?

colloqi, 2024-05-12:
Yes, that is preferred and also if the location of login does not change, there is no need for any authentication as well

Buzzmovies, 2024-05-12:
In other words, it's not a bug, it's a feature 😅
Thank you for the explanation. Have a good day.

closed 2024-05-12 by Buzzmovies