colloqi/piSignage #260: How to send otp for API login?

miroppb, 2024-05-12:
With the new change forcing OTP, how can I include it in my api call to `/session`?

Original body:
`{
  "email": "swagger",
  "password": "swagger@piathome",
  "getToken": true
}`

New Response:
`{
	"message": "OTP needed",
	"provider": "ga-otp"
}`

I tried adding the key "otp" and "ga-otp" to the body, but neither one seem to be the correct one.

colloqi, 2024-05-13:
1. Send a POST request to /api/session with your ( email, password, getToken ) in the body of the request. You should receive a response with 401 Unauthorised along with an OTP to your registered email.
2. Now send a POST request to /api/session with the (email, password, getToken, code : <received-otp-code> ) in the body of the request. You should get a 200 OK response with the login token.

JonasJes, 2024-05-13:
Shouldn't this be updated in the Swagger documentation? https://piathome.com/apidocs/#/session/login

closed 2024-05-13 by miroppb