colloqi/piSignage #262: pIsignage on wifi cannot connect

NathanD19, 2024-05-14:
Using raspberry pi 5 & version 5.1.4 we cannot connect to the signage portal on the Pi over Wi-Fi after an arbitrary amount of time.
Only when we restart the device, we can access the the device signage portal.

Thanks!

AbhijithBn, 2024-05-15:
@NathanD19 
Just to confirm, is the PlayerUI at http://<player-ip>:8000 the signage portal that you are referring to? 

Are you able to login to the player via SSH when this happens?

NathanD19, 2024-05-15:
Correct, the Player UI on 8000

After change the location of the device, it seems to connect properly and not drop out. Not exactly sure what happened here.  Could be network issues my end.

AbhijithBn, 2024-05-16:
If it happens again

Please SSH into your Pi and send us the output of `nmcli connection show` to support at pisignage.com