colloqi/piSignage #266: does not work on a Raspberry Pi Zero 2 W

Lukas1633, 2024-06-24:
Hello,

I have a problem getting PiSignage to run on a Zero 2W. The installation goes through without any problems. I follow the instructions and flash version 3.2.9 onto the SD card. No settings are set by the imager. WLan settings so far also no problems. If I want to add an asset, this also works. There are also no problems when creating a playlist. 
When I want to add assets to the playlist, they are not displayed. I can still navigate to the home screen, but the memory is no longer displayed and I can no longer select a playlist. The assets created a few minutes ago are also no longer displayed under the assets. If I then refresh the web page, it shows that the host is not accessible. The Raspi can still be reached via Putty and ping.

Thank you very much for your ideas!

colloqi, 2024-06-27:
Hi

Can you check logs forever_out and forever_err for any crashes and let us know?

Also you could see the responses under the inspect -> network tab of chrome

Thanks

Lukas1633, 2024-07-01:
hey,

I can only download the two log files as long as the web interface has not yet crashed. I downloaded them after the restart.
[forever_err.log](https://github.com/user-attachments/files/16053214/forever_err.log)
[forever_out.log](https://github.com/user-attachments/files/16053216/forever_out.log)

I don't quite understand the second way. Do you mean the Chrome developer tool? If so, this is displayed when I create an asset. No matter what type of link display I specify, this error message is displayed.
[image]

Lukas1633, 2024-07-02:
Here are the Log files after the chrash.
[forever_err.log](https://github.com/user-attachments/files/16066372/forever_err.log)
[forever_out.log](https://github.com/user-attachments/files/16066373/forever_out.log)

patriot1889, 2024-08-05:
Same problem with a raspberry pi 2.