colloqi/piSignage #269: Rotate Video Output?

makru83, 2024-07-08:
We installed pisignage 5.2 on a rpi4, but the Display attached is mounted uside down.
Is it possible to rotate the Output on the rpi through pisignage by 180°?

The Monitor unfortunatly doesn't have the feature.

AbhijithBn, 2024-07-09:
Hi

If you are using piSignage v5.x.x, can you try using the Vertical flip option from Group Settings?
[image]

https://help.pisignage.com/hc/en-us/articles/360001234431-Group-Settings-

slsimic, 2025-07-09:
That "solution" would rotate all screens in the group, isn't it? And most probably not all screen in the group need to be rotated. Which lead us to another "solution" to make group for rotated screens only, but there can be three groups for each 90 180 270 degrees of rotation. Instead of just make setting per player which would be simplest and easier.

hitesh83, 2025-07-10:
I was in same situation when we had install 10 screens, and one screen was mounted up side down and was in a cabinet which is 15 ft up from ground.
So i SSH the related Pi and used config.txt after all player setup is done from server.
this might work for you.
This settings will get overwritten by server next time you change group settings in server.

davidecat, 2025-07-10:
To create a new group is "the solution".
A workaround could be a bash script to "watch" the config.txt file and run a sed command to replace the orientation everytime the file change. Be careful this could trigger an infinite loop :-)

slsimic, 2025-07-14:
No, creating a group for that is not a solution, because Player can belong only to one group. Tried it.
So you cannot have say, 8 players grouped by location and then have 2 that are in portrait mode. I didn't find this achievable with PiSignage.