colloqi/piSignage #270: PiSignage Open Server layout issues

GregoryWest501, 2024-07-20:
I am running PiSignage open server.  The biggest reason for this is I want to display a couple camera feeds that are NOT accessible on the internet.   The issue I am having is the layout editor on PiSignage server I can not find where to adjust Zone 5 and Zone 6.    Also when I made changes to the first 4 zones (Main, Side, Bottom, Zone 4) only the changes to Main are getting to the player.

Is this a known issue, if so is there a work around?

Greg

davidecat, 2024-07-21:
Hi, i leave the answer your main question to other, I write only to suggest that is not necessary that the server reachs the local network cameras to work, its enough that the local players can reach. 
Regards

colloqi, 2024-07-21:
Greg, you could use the template designer at pisignage.com (login to your account) and give the sizes to all zones. Then download the template and upload as asset to your server to use in the playlist. Make sure to clear the values under layout popup.