colloqi/piSignage #273: Youtube subtitles

istoutjesdijk, 2024-09-06:
Hi,

When I add a Youtube link the player selects English subtitles by default, is there a way to disable subtitles or change the language of the subtitles?

hitesh83, 2024-09-06:
just add '&cc_lang_pref=fr&cc_load_policy=1' to the video's embed URL.

fr is for french 
Likewise you can add your desired language.

istoutjesdijk, 2024-09-06:
I tried this with both regular and embed URL's. When I use them directly in the browser it works, but not with the players.

hitesh83, 2024-10-24:
Workaround -
Make a html file to load youtube embedded video in browser, set size auto play subtitles etc in html
And schedule the html file which will load YT in the players browser 
(Not the correct way but works for me every time)