colloqi/piSignage #278: This player is powered by pisiganage.com ticker is coming

SarathKumarMB1833, 2024-10-23:
i am using the palyer only open source licence for 2 free player for a local host and when i load the content the ticker is coming.How can i solve this

hitesh83, 2024-10-24:
Register your players ID on pisignage website in your account.
Download Licence txt files.
Upload them in open source server, keep username in settings same as your pisignage account.

AbhijithBn, 2024-11-29:
Hi @SarathKumarMB1833  

Did you check the points mentioned in this article https://help.pisignage.com/hc/en-us/articles/360001256851-This-player-is-powered-by-pisignage-com?

Especially,

- license should be generated as "Player only" while registering
- installation name should be same as your username at [pisignage.com](http://pisignage.com/) (not email ID)
- license file name should be same and without any extra characters like ~1
- License file is uploaded to your server and player is online

DorkSaber, 2024-12-12:
@AbhijithBn Yes I second @frankchau93, I'm accessing my open source server via a Cloudflare reverse proxy, and I'm also getting domain not matching. How do I resolve this? Is there somewhere in the installation I can input the reverse proxy domain? I really don't want to use port forwarding. And even if I did, would that even work? Or do I have to have them on the same lan? There must be a way. @frankchau93 did you figure it out?

hitesh83, 2024-12-12:
the domain here is not referred to your reverse proxy, it might be referring to **UserID**.pisigange.com, this **UserID** need to mentioned in your opensource server's setting.
[image]