colloqi/piSignage #279: Player crashes when i try to modify a playlist

Atomhoernchen87, 2024-10-23:
Im using a Raspberry pi 3 model B with the latest update for this player. 
I can upload assets and manage options, all works fine. 
When i try to access a playlist, the assets are not visible. If switch back to assets, they're gone too. If i reload the ip, theres no connection.

Im not using cloud or something, just a laptop to get the files locally on the player. 

What can be the reason for this strange bug?

thanks in forward!

colloqi, 2024-10-24:
Server changes override player configuration, can you change the server name to something non-existing like pisignage1.com and try?

Atomhoernchen87, 2024-10-24:
Still not working. I can't even assign an asset to a playlist. If I try to assign nothing happens and theres nothing to see at playlists next to the asset.

hitesh83, 2024-10-24:
Few suggestions 
1- write the image again, let it boot and change the server name to unreachable address and reboot. 
After this upload files make a playlist and play it from webUI.

2-  try going back one or two versions, new versions were mostly helpful for websocket server updates which you are not going to use.

_3- this is totally unofficial unauthorised should not be mentioned here. If you are using a player only to play a few mp4 files locally without a server, then you are not using pisignage properly and not utilising all capabilities, your desired results can be archived by scp and with couple of 4-5 lines of scripts in the shell._ (admin - please delete this if not as per guidelines)

Atomhoernchen87, 2024-10-29:
Where can i get the older versions? Can´t find them.

akorchuk, 2024-12-08:
I have the same issue.
For the test I'm not using server (no WiFi connection was set up)
Downloaded image pisignage_3.2.9.img written to flash using RaspberryPi Imager
After connecting to PiPlayer_XXXX hotspot I've uploaded a movie. It's played when I press Play from Assets screen. 
But when I assign it to any playlist webpage stops responding.