colloqi/piSignage #284: Zero 2 W lagging or getting stuck

smitterer, 2025-02-02:
Hi!

I'm using pisignage 5.3.4 on a pi zero 2 W.
Unfortunately it is lagging or stopping completely when playing videos.

I tried the following things:
Set resolution to 720 and 1080
Used a stronger power supply
Set letterbox instead of stretch

Is the pi zero 2 W not capable running videos? I didn't see any information on that.

Thanks

colloqi, 2025-02-03:
Hi

You could set VLC under Group settings for hardware assisted play which should work. Otherwise you can always use 3.2.9 which uses earlier omxplayer

smitterer, 2025-02-03:
Ok thanks.
Setting VLC didn't work but downgrading to 3.2.9 did.

Unfortunately the logo is shown with several lines in it and it is greyed out (see the image)
Does it have a problem with transparent pngs?

[image]