colloqi/piSignage #286: Disappearing license code

Nevov21, 2025-03-11:
Hello, Ive got my license file via email and I have it under /home/pi

[image]

but after restart a device the number in the file is dissapering.

Ive downloaded pisignage version 3.2.0 and im running it on the Raspberry PI 3B+.
Greetings
Thanks for answer.

Nevov21, 2025-03-11:
And the log says:
3/12/2025, 3:06:31 AM - 8000: error: license file wget result: --2025-03-12 03:06:31--  http://192.168.1.18:8000/licenses/admin/license_000000000b83aea8.txt
Connecting to 192.168.1.18:8000... connected.

Can i change the options somewhere to change the path where the program is looking the license?

hitesh83, 2025-03-12:
You need to upload these license txt file in your opensource server under settings.

Pi will auto download the license file from server at startup.

Manually copying license file to pi is not the solution.