colloqi/piSignage #287: Function requested

hitesh83, 2025-03-12:
Option to remove/ add  assest from/to all playlists by going to assets property.

Assume I am having a XYZ.mp4 
With this function we can add this xyz.mp4 to selected playlist in one go or remove it from selected playlist in one go.

Also we will get to know where all xyz.mp4 is used (in which playlist)

AbhijithBn, 2025-04-07:
Hi @hitesh83 

You should be able to Add/Remove an asset from a Playlist from the Asset Details Page.
It also shows which playlist the asset is already a part of.

hitesh83, 2025-04-07:
It is in which release? I think I am on latest one [5d0aba9] , I see no such options.
Please help me find it.  
Attaching screenshot of asset details page in current setup.

[image]

AbhijithBn, 2025-04-09:
Hi @hitesh83 

Sorry for the confusion. This feature is limited to pisignage.com for now. We'll let you know if we plan to integrate this to the open-source server.