colloqi/piSignage #291: Player Update/Upgrade offers lower version

LazyInLA, 2025-05-05:
My players are running 4.8.1, I would like to upgrade for some bugfixes.  The Software Update button in the open-source server offers to update from the current version to 3.3.0.  Can I not upgrade to the 5.x series with this method?  Is manually writing the image to the micro-sd card the only way?  If so, will it invalidate my license files?

Thank you for your time

colloqi, 2025-05-06:
Hi, is your server version upto date with github revision. 4.8.1 being old version suggest SD card rewrite with 5.3.2 to get the latest Raspbery OS 
Same license will work, otherwise write to support to redeem the old license.

LazyInLA, 2025-05-06:
Server is on latest version as far as I can tell, I have done the git pull origin master/npm install procedure.  I will try the SD card method.  Thank you for your help.

closed 2025-05-06 by LazyInLA