colloqi/piSignage #297: Web pages reloading on every asset switch

denyskon, 2025-07-29:
Hello, we're using licensed PiSignage players with an open-source Server to display Grafana dashboards. However, even though caching is configured in group settings, on every asset switch/every round of the playlist the pages are being reloaded completely, causing a delay of up to 10 seconds until the page is visible. This happens with both webpage and weblink asset type.

Is there any fix for this?

ib0rik, 2025-09-07:
Hi,

I am in the same situation.

Using weblinks/webpages, I want to set up a playlist switching static web assets every 5 seconds.

But the webpages are reloaded every time we switch to a web link asset. I guess the browser or tab is destroyed when not shown and created when shown.

It is not useful from my case because reloading takes time (2.5 seconds) and it is disturbing for my signage's viewers.

I tried to activate the "Keep webpage in memory" setting and deactivate the "Reload link URLs each time" setting but still webpage and weblink assets keep reloading.

Is it not possible to choose if a web asset should be loaded only once and shown as is for the next rounds ?

Thank you.

ib0rik, 2026-04-17:
I have 2 weblink assets switching every 10 sec in my playlist. So no video player involved at all.

My both weblinks point to a static content that when you call it for the first time you have a loading bar or you can call it a welcome splash screen before to fade out and show the main content. And everytime the piSignage player switch assets to another one (all are weblinks) I see this "welcome" loading bar.

Maybe the Chromium browser is destroyed everytime it is not the displayed asset or anything else but it is clear that the web page is always (re)loaded when it is displayed on screens. It is not a visual impression but a fact in my point of view.